Imroy/pubsubclient #1: Compile error

Toshik, 2015-05-14:
```
PubSubClient.h:55:56: error: section attribute not allowed for 'payload'
```

for that code:

```
   bool publish_P(String topic, const uint8_t PROGMEM *payload, unsigned int, bool retained = false);
```

Toshik, 2015-05-14:
I think it is related to https://github.com/esp8266/Arduino/commit/8b1caebf7689ec419ff14e9d966e0929de1a67a3

Imroy, 2015-05-14:
696e1a5 hopefully fixes things. I haven't been able to test this because my only usable board is currently being used to test battery life.

Toshik, 2015-05-14:
@Imroy I will test now

igrr, 2015-05-14:
Will this still work on AVRs?

Toshik, 2015-05-14:
@Imroy Now it compiles.

Imroy, 2015-05-14:
@igrr I pull in ESP8266WiFi.h to get the WiFiClient class. Actually, I could just pull in WiFiClient.h. But in AVR land that's in some optional library, isn't it?
That's the only external dependency that I know of, besides the core Arduino headers. Otherwise it should work on AVRs.

closed 2015-05-18 by Imroy