Imroy/pubsubclient #2: undefined reference to `_sbrk_r

gnih, 2015-05-14:
Hi,

I encounter "undefined reference to `_sbrk_r" error with arduino ide 1.6.1 and 1.6.3

Missing hard coded path?

[image]

Imroy, 2015-05-14:
This is a problem that has been fixed in the Arduino-ESP8266 port (see esp8266/Arduino#11), but there is no recent build. The maintainer(s) is hoping to get a new release out this week or next week (esp8266/Arduino#188). Until then, you can build your own version from Git (what I do) or maybe try Appveyor - https://ci.appveyor.com/project/igrr/Arduino.

gnih, 2015-05-14:
Thank you for your quick reply!
Have a nice day,
Kind regards,
G.

jeffryr, 2015-05-14:
It also works with Arduino-1.6.4 plus the Adafruit ESP8266/Huzzah plugin.  I suspect it would work with Sandeep Mistry's addon, I see he now has a 0.0.5 release but I haven't personally tried that one.  I do all of my builds on Linux BTW, don't know what the status is on Windows or OSX.

closed 2015-05-31 by Imroy