Imroy/pubsubclient #3: No heap available, failed to malloc 0

gnih, 2015-05-15:
Hi,

Notre sure it's an issue because it seems to work anyway but I get this error every 15 seconds when using the basic example (I didn't try the others...).

Anyone else facing this error?

Arduino 1.6.4 with esp package from Adafruit 1.6.2

BJvDL, 2015-05-17:
Exactly the same issue here. i'm also using Arduino 1.6.4 with esp package from Adafruit 1.6.2
Maybe it comes from other mqtt clients publishing topics to the same broker but where this client is not subscribed to?

Imroy, 2015-05-18:
I'm not seeing this problem. It may be caused by a problem that has since been fixed in the Arduino-ESP8266 port. AFAIK, Adafruit's builds are about a month old by now.

BJvDL, 2015-05-19:
I upgraded to arduino 1.6.4 and the latest version from Ivan Grokhotkov (https://github.com/esp8266/Arduino) via boardmanager and the problem still persists

gnih, 2015-05-19:
Worst here because it doesn't even compile anymore... :-(

[image]

[image]

BJvDL, 2015-05-19:
Did you uninstall the Adafruit version first?
Delete the link in Preferences, reload arduino and load the board manager to see it has gone.
After that put in the new link from the ESP8266 community, reload arduino and install it in the Board Manager again. Maybe both versions get mixed up in the background

CptanPanic, 2015-05-19:
Also check the latest esp-arduino as they added support for board manager directly. Should be most recent.

BJvDL, 2015-05-19:
I use  http://arduino.esp8266.com/package_esp8266com_index.json and that one is always the latest

gnih, 2015-05-19:
Ok after complete fresh install : uninstalled arduino ide & manually deleted all arduino files and, last but not least, deleting the pubsub library in documents folder (how could it be altered!??).

So now it compiles, let check for the no free heap message...!

Imroy, 2015-05-20:
@gnih That "PROGMEM" compile error was reported in #1 and fixed. Updating pubsubclient might have been all you needed.

BJvDL, 2015-05-23:
I'm still having 'No heap available, failed to malloc 0' .Arduino IDE is 1.6.4 with the latest version of ESP8266 in the Board Manager and Pubsubclient is up to date. I'm using pubsubclient-master.
There is also a pubsubclient-oop. What is the difference?

gnih, 2015-06-03:
Same here with latest version.
The strange thing is that by using
Serial.print("free heap="); Serial.println(ESP.getFreeHeap());
in the main loop, I always get >32k free heap but the debug still prints
No heap available, failed to malloc 0

Any idea?

igrr, 2015-06-03:
```failed to malloc 0``` means that someone tried to malloc 0 bytes. While malloc(0) is certainly legal, is seems to be not supported well by the underlying implementation.

Imroy, 2015-06-03:
I've added checks in a few places that use malloc() and could be called with a value of zero. See if this fixes your problem.

igrr, 2015-06-03:
Actually i thought to fix that in our malloc() so that it would guard against this error in pvPortMalloc. But handling this in the library is also okay.

igrr, 2015-06-03:
Also i see some places in 47cfda7 where you guard against free(NULL), which should actually work.

Imroy, 2015-06-24:
@gnih @BJvDL Can this be closed now?

BJvDL, 2015-06-24:
@Imroy , I tested with your latest library and no problems anymore.
Please close this issue.
Thank you

closed 2015-06-24 by Imroy