Imroy/pubsubclient #4: Change the IP address

sticilface, 2015-05-16:
Hi,

I've started using your fork as the older version no longer compiled on the ESP arduino IDE. 

I have got it working apart from one thing, I can't assign the IP address dynamically.  I want to read the IP address from the EEPROM then start the MQTT based on this.   When i define MQTTserver as below it all works... 

```````````
      IPAddress MQTTserver(192,168,1,24);
      PubSubClient mqttclient(MQTTserver);
`````````````

If i do not, and try to assign MQTT server from EEPROM... during set up... I get the right IP but it fails to connect...
`````````
     for (int i=0; i<4; i++) {  
    MQTTserver[i] = EEPROM.read( mqttAddress+i);
   }
`````````````````

Any ideas why? I had this working before when the IP address was a char array..

Imroy, 2015-05-16:
Ah, I see. The server hostname/IP address is only used in connect(), so it should be safe to set/change it before then.

Imroy, 2015-05-16:
With this update you create your global client object:
````
PubSubClient mqttclient;
````

And then later assign the server ip address:
````
mqttclient.set_server(MQTTserver);
````

sticilface, 2015-05-16:
haha... I've just been modifying the example script to show that it doesn't work and you've come up with a solution!  fantastic.  Thank you!

Will try it out

sticilface, 2015-05-16:
Works perfectly.  Thank you...  I've been going bonkers trying to get that to work today...

closed 2015-05-18 by Imroy