Imroy/pubsubclient #8: Different behavior between DHCP and fixed IP when WiFi drops out and returns

BJvDL, 2015-05-23:
I'm testing, testing, testing... with the example mqtt_basic to see the behavior with a bad wifi connection.
My goal is to get the MQTT connection always to be restored after a Wifi drop. My ESP should NOT restart. It is doing other tasks also and they should not be disturbed. MQTT is only used to get some commands and write some status.

My first test is the mqtt_basic example as is. It responds to the subribed item as expected.
Now I switch the AP off and on again and see what happens: No problems with WDT, but it is not responding to the subscibed item anymore. -> The connection to the broker is lost.

Now I bring in some code in void loop() to restore the subscription:

unsigned long previous_time = millis();
void loop()
{
  bool mqtt_status=client.loop();
   if(!mqtt_status)
   {
        if (client.connect("arduinoClient"))
        {
       Serial.println("Connecting to MQTT broker....");
       client.publish("outTopic", "hello world");
       client.subscribe("inTopic");
       }

unsigned long current_time = millis();
  
  if ((current_time - previous_time) > 1000) // print debug information every second
  {
    previous_time = millis();
    Serial.println("loop"); // check if void loop() is still running
   if (mqtt_status)
    {
      Serial.println("MQTT connected");
    }
    else
    {
      Serial.println("MQTT fail");
    }
  }

}

This fires the WDT after a WiFi loss and sometimes after a WiFi restoration. Sometimes it not even reboots. I sometimes get FatalException(9) and FatalException(28), but that is random i guess.

My third test is to set the IP to fixed. Now there are no WDT problems anymore. it reconnects to the broker again and reponds to the subscribed item. I can see the timeout in the library. When my WiFi AP is switched off it still prints "MQTT Connected" for a while and after the timeout period it will print "MQTT fail". switching the AP on again the ESP restores the WiFi connection and after that the code restores the MQTT connction. This is how it should work and it does.
But now it keeps connecting to the MQTT broker every 10-15 seconds.

My conclusion is that the ESP doesn't like DHCP with the subpubclient if WiFi gets lost. It looks like there is no network adapter anymore and it crashes. Anyway it has different behavior.

Testato, 2015-06-20:
I simply check befoure every publication if wifi is connected and if client is connected.
- if i stop wifi (restart my router), the ESP reboot whit a WDT error. After that when the router reboot the connection automatically reconnect OK
- If i stop mosquitto daemon, when I restart it, the connection automatically reconnect OK whithout WDT

BUT:
if i stop the broker connection, and after i stop also the wifi connection, the ESP8266 freeze, no WDT error, only freeze whithout any activity. I must reset manually the ESP.

May you try this scenario also ? this is worst that the auto WDT reboot, because you must reset or remove power to the module.

I will try whit fixed IP on ESP

av1024, 2015-06-22:
@Testato Confirmed partially.
Looks like there is wifi error. My ESP resets with 'beacon timeout' and by WDT. BUT! It freezes after wdt reset, not starting my code. Also I have constantly rebooting ESP while wifi router is off using arduino git build 61d9b1f4 . There is no reboots on older arduino version (git from May 28)
Log 1 (conditions same as yours):
```
* MQTT Publish sensors   2015.06.22 09:55:50.9
debug on
setDebugOutput to on
OK
esp-9e82d0> 
beacon timeout
rm 0
pm close 7 0 0/3921100541
Fatal exception (28):
epc1=0x4020918a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
After hardware reset:
```
scandone
Fatal exception (28):
epc1=0x402090ce, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

igrr, 2015-06-22:
Latest version should have a stack trace printed to console — can you please post that and your .elf file as well? Thanks.

BJvDL, 2015-06-24:
Using version 1.6.4-835-g77d77e8 seems to have the problem solved.
Now the sketch uses DHCP and I can switch off the router without the WDT messing up.
After restoring the WiFi the ESP gets it's IP again and connects to the broker again to continue publishing

I can't post .elf files here, but here is the console

Publishing...
Publishing...
Publishing...
rm 0
pm close 7 0 0/74277806
reconnect
f -240, scandone
no Zonder_draad found, reconnect after 1s
reconnect
f 0, scandone
no Zonder_draad found, reconnect after 1s
reconnect
f -240, scandone
no Zonder_draad found, reconnect after 1s
reconnect
f 0, scandone
no Zonder_draad found, reconnect after 1s
reconnect
f -240, scandone
no Zonder_draad found, reconnect after 1s
reconnect
f 0, scandone
no Zonder_draad found, reconnect after 1s
reconnect
f -240, scandone
no Zonder_draad found, reconnect after 1s
reconnect
f 0, scandone
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with Zonder_draad, channel 3
dhcp client start...
ip:192.168.12.106,mask:255.255.255.0,gw:192.168.12.254
Connecting to MQTT broker (192.168.12.3) as 18:FE:34:A0:C5:77
Connected to 192.168.12.3
Publishing...
Publishing...
Publishing...

Testato, 2015-06-24:
here also solved whit 1.6.4-835-g77d77e8
- No wdt reboot after wifi router switch off
- No freeze if i stop brocker and after switch off the wifi router
Thanks @igrr

Lstt2005, 2015-07-08:
The same problem! Version is 1.6.5-804-g2d340c7 - lastest...

Testato, 2015-07-09:
nope,
your version is 1.6.4, 1.6.5 do not exist (it is a tag error)
For use the g77d77e8 build you must use the development .json link

Lstt2005, 2015-07-12:
No error, in file http://arduino.esp8266.com/staging/package_esp8266com_index.json -we see
"url":"http://arduino.esp8266.com/staging//esp8266-1.6.5-804-g2d340c7.zip",
      "archiveFileName":"esp8266-1.6.5-804-g2d340c7.zip",
and this version is istal from internet for all at this time...(

Testato, 2015-07-12:
g2d340c7 is the actual developmen version (the 1.6.5 number become from arduino repository, it is not important here, @igrr can clear this point)
On g2d340c7 i do not have problem