Imroy/pubsubclient #9: Can't pass 'this' to callback scope

jmas, 2015-06-08:
Hello. I'm not super-c++ programmer, but I found that I can't pass 'this' to lambda-function when I work with `set_callback()`. I want to do something like this:
```cpp
        mqttClient.set_callback([this](const MQTT::Publish& pub){
            for (int i=0; i<eventsSize; ++i) {
                if (pub.topic() == events[i].name) {
                    events[i].callback(pub.payload_string());
                }
            }
        });
```
`eventsSize`, `events` is members of class scope. So I did some modifications for `PubSubClien.h`:
```cpp
//typedef void(*callback_t)(const MQTT::Publish&);
//...
std::function<void(const MQTT::Publish&)> _callback;
//callback_t _callback;
//..
PubSubClient& set_callback(std::function<void(const MQTT::Publish&)> cb) { _callback = cb; return *this; }
```
And now I can pass lambda-function to set_callback() with different scope.

I can do Pull-Request with this modifications. But if you know right way to pass 'this' to lambda-function - please tell how I can do this.

Imroy, 2015-06-08:
It's simpler to just redefine callback_t using std::function.

jmas, 2015-06-08:
@Imroy, Can you show some code how to do this. I'm noob in this. :(

closed 2015-06-08 by jmas

Imroy, 2015-06-08:
Okay, good to hear you figured it out :)