Imroy/pubsubclient #13: maximum length of message

sticilface, 2015-06-20:
What is the maximum length of message? 

I'm getting quite a truncation at about 50B, when i can send massive messages using MQTT-SPY

sticilface, 2015-06-20:
here is what i get from serial...
```
MQTT msg SENT: speed/effectlist, Message: effect:Off,Rainbow,Color,Chase-Rainbow,Fade,AdaLight,test,Looparound,Pick-Random,Fadein-FadeOut,Cool-Blobs,UDP,Rainbow-Cycle,Spiral,Squares2,Squares-Random,Test4,Squares,EQ1,Random-Colour,Random-Function,Art-Net,HSV-Top-Bottom,Linear-Top-Bottom,Single-Colour-Fade,Random-Colour-Fade,HSIcycle,New-Animations
MQTT Message Recieved: speed/IP => 192.168.1.162
MQTT Message Recieved: speed/Version => WS2812
MQTT Message Recieved: speed/Status => Device Ready
MQTT Message Recieved: speed/effectlist => effect:Off,Rainbow,Color,Chase-Rainbow,Fade,AdaLig
```

the string that is passed to the message send is shown msg SENT... but what is received is much sorter.  i can confirm that this is the case using mqtt spy.  i can send longer messages using mqtt spy.  i've upped the max message size to 256 in MQTT.h but that has not helped.  any ideas?

Imroy, 2015-06-20:
Hmm, increasing MQTT_MAX_PACKET_SIZE in MQTT.h should have allowed slightly larger packets. It's a left-over from knolleary's original code. Also, many buffer sizes and positions are stored in 8-bit integers. So that's another limit there.

I'm working on a more dynamic solution that should allow up to the maximum of 256MB packets. Of course, the ESP8266 doesn't have anywhere near that much RAM.

sticilface, 2015-06-20:
Ah that would be fantastic,  i did notice all the code was in uint8_t....  
I'm using it as a way of auto populating a drop down list of choices in my home automation software.  This way you don't have to do anything to update things when you've made changes.  I've got everything in place, but being limited to 50 or so characters is not enough.  

Solid library though.  keep up the good work.  Let me know if you want me to test anything

Imroy, 2015-06-20:
I've pushed my changes and my little sensor node still seems to be working. I'm going to bed soon so you have a while to test it before I'm available again :)

sticilface, 2015-06-20:
Im about to head off as well.. so it will be tomorrow.  but thank you!

sticilface, 2015-06-20:
only kidding.  just tried it out, and it has published the whole message, so that has worked.  Thank you.

Testato, 2015-06-21:
this is a great improvement.
Thanks to stilface for the discovery, and Imroy for the solution :-)

closed 2015-06-22 by Imroy

andreshidalgo08, 2016-09-01:
Hi, I'm new to the use of ESP and MQTT, I'm currently working on a NodeMCU v1.0. I would like to know what is the new limit length for the messages, I'm currently able to receive a 1011B payload using  Mosquitto broker but I'm aiming for a 40MB payload, is this possible?

My code:

`#include <SPI.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

const char *ssid     = "xxxxx";
const char *pass = "xxxxxxx";

IPAddress ip(192,168,200,60);               // Blue 192,168,200,60 White 192,168,200,70
IPAddress subnet(255,255,255,0);
IPAddress gateway(192,168,200,1);

IPAddress server(192,168,200,80);            //Broker MQTT

WiFiClient wclient;
PubSubClient client(wclient, server);

#define BUFFER_SIZE 150000

void callback(const MQTT::Publish& pub) {
  Serial.println("callback");
  /*
  Serial.print(pub.topic());
  Serial.print(" => ");
  if (pub.has_stream()) {
    Serial.println("Aca2");
    uint8_t buf[BUFFER_SIZE];   //Inicializa el buffer
    int read;                   //Inicializa un int para
    Serial.println(pub.payload_len());
    
    while (read = pub.payload_stream()->read(buf, BUFFER_SIZE)) {
      Serial.write(buf, read);
    }
    pub.payload_stream()->stop();
  } else {
    //Serial.println(pub.payload_string());

    //Serial.println(pub.payload_string()[0]);

    Serial.println("Aca");
  }
  */
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  
  pinMode(LED_BUILTIN, OUTPUT);

  //Inicializacion para comunicacion SPI
  pinMode(D1, INPUT);   //output from display /TC_BUSY (if 0=busy  if 1=ready to receive new commands)
  pinMode(D2, OUTPUT);  //TC_EN (1=disabled 0=enabled)

  digitalWrite(D2, HIGH);
  Serial.println("/TC_EN: ");
  Serial.println(digitalRead(D2));
  
  digitalWrite(D2, LOW);  //to enable /TC_EN
  Serial.println("/TC_EN: ");
  Serial.println(digitalRead(D2));
  
  SPI.begin();
  pinMode(SS,OUTPUT);
  
  digitalWrite(SS, HIGH);
  Serial.println("/TC_CS: ");
  Serial.println(digitalRead(SS));

  delay(1);
}

void loop() {
  // Para conectar al WiFi
  if (WiFi.status() != WL_CONNECTED) {
    Serial.print("Connecting to ");
    Serial.print(ssid);
    Serial.println(".");
    WiFi.begin(ssid, pass);
    WiFi.config(ip, gateway, subnet);

    if (WiFi.waitForConnectResult() != WL_CONNECTED)
      return;
    Serial.println("WiFi connected");  
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
  
  if (WiFi.status() == WL_CONNECTED) {
    if (!client.connected()) {
      if (client.connect("Blue")) { //Id del dispositivo
  client.set_callback(callback);    //callback, que va a hacer cuando reciba un mensaje
  client.subscribe("demo/board2", 2);   //topic al que esta subscrito, usar una variable
      }
    }

    if (client.connected())
      client.loop();      //Wait for packets to come in, processing them.
      Serial.println("loop");
  }

  //Para verificar que esta funcionando
  digitalWrite(LED_BUILTIN, LOW);   //Enciende led
  delay(200);
  digitalWrite(LED_BUILTIN, HIGH);  //Apaga led
  delay(200);
}`

When I send a 1012B message I get this exception:

Exception (9):
epc1=0x40203dc4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000005 depc=0x00000000

ctx: cont 
sp: 3ffef590 end: 3ffef820 offset: 01a0

>>>stack>>>
3ffef730:  3ffee7a4 0000019d 00000000 40203a1e  
3ffef740:  00000000 00000000 00000000 4010068c  
3ffef750:  00000000 00000001 3ffe88f5 3ffee7f0  
3ffef760:  3fffdad0 00000001 3ffee650 40203c46  
3ffef770:  40105244 00000001 3ffee650 40204020  
3ffef780:  4021c21e 3ffee7e8 3ffee6d4 3fffdad0  
3ffef790:  4021d073 3ffee7f0 402048d4 3ffee800  
3ffef7a0:  3fff012c 402024d0 000000c8 40204da8  
3ffef7b0:  3fffdad0 00000001 3ffee650 40201e81  
3ffef7c0:  00000000 00000000 00000000 00000000  
3ffef7d0:  40201be8 00000000 40204afc 40204ae8  
3ffef7e0:  40201be8 00000000 3ffe88f0 01c8a8c0  
3ffef7f0:  00000000 00000000 00000001 402048f5  
3ffef800:  3fffdad0 00000000 3ffee7e8 40204920  
3ffef810:  feefeffe feefeffe 3ffee800 40100718  
<<<stack<<<

Since I'm new to this I don't really know what to do to solve it.
Great work, thanks for the library!