Imroy/pubsubclient #14: MQTT work only on string ?

Testato, 2015-06-20:
May you confirm this ? MQTT work only on char/string ?

I tried to send the value of a numeric variable, by the constructor:
    bool PubSubClient::publish(String topic, const uint8_t* payload, unsigned int plength, bool retained)

but when i send number from 0 to 255 (uint8_t*) on mosquitto the number is converted to Ascii Character.

So whit this library (all MQTT library ?) we must send always String ?
I must use itoa_function for publish a numeric variable ?

Imroy, 2015-06-21:
Like this?
```C++
  uint8_t value = 7;
  client.publish("data", &value, sizeof(uint8_t));
```
I get this:
```
$ mosquitto_sub -h 192.168.2.2 -t 'data' -v | hd
00000000  64 61 74 61 20 07 0a
```

Testato, 2015-06-21:
yes i used some like this but on mosquitto side i used
``` 
mosquitto_sub -d -t home/temperature  
```

So depend on whitch tipo of reading option we use ? i must study the mosquitto option ?

The `unsigned int plength` is the size of the value ?

closed 2015-06-26 by Testato