Imroy/pubsubclient #15: The delayMicrosecond(100) in PubSubClient::wait_for causes wdt and/or exception failure

av1024, 2015-06-22:
The `delayMicoroseconds` does not update WDT so I have lock-ups, wdt or exception failures on stress-tests (stop MQTT broker, turn off WiFi etc).

Look like replacing in by `yield()` solves an issue.

Testato, 2015-06-22:
the Arduino esp8266 core document that delayMicroseconds() stop the wifi stack, but the delay() not. So maybe we can try whith delay(1), 1millisecondo instead of 0.1.
`delay pauses the sketch for a given number of milliseconds and allows WiFi and TCP/IP tasks to run`

whit the yeld() you removed completely the delay because it is like delay(0)

Imroy, 2015-06-22:
Yes, I had forgotten that delayMicroseconds() doesn't call yield. What if yield is called right after?

Testato, 2015-06-22:
Probably the best is use a millis timing whithout use of delay, not only here but in all the library ?

av1024, 2015-06-22:
Why do you need 100us delay here? Is it 1ms me enough?
Or simply use yield() (equals to delay(0)) and millis()

av1024, 2015-06-22:
@Testato yield() is enough because it used for switch scheduler, but time interval measured by millis()

Imroy, 2015-06-22:
I'm not sure why I chose 100us. It doesn't seem to be inherited from knolleary's original code, so I can't blame him :P
I think I just didn't want it to spin its wheels checking a simple condition and to have a small delay in there. But I forgot about yielding. And it worked for me(tm).

av1024, 2015-06-22:
> And it worked for me(tm).

For me too, But not for latest git Arduino's 8) I have got a mysterious resets on not clean wifi area but I hope they are gone...

igrr, 2015-06-22:
Guys, this code was there since original Nick's version. I remember it caused problems so I had a workaround in my code which called yield() internally if it detected that you are polling WiFiClient::available() in a loop. It seems to be broken now — I will check that.

My suggestions is to replace delayMicroseconds(100) with yield();

closed 2015-06-23 by av1024