Imroy/pubsubclient #16: MQTT - knolleary or ESP8266 library not working in same directory

questuk, 2015-06-22:
Hi Imroy,

I am trying to use MQTT using the following:
Ethernet Arduino board or using ESP2866 module

For the past 3 years I have been using this library, https://github.com/knolleary/pubsubclient this has worked well.

I now wish to use the ESP8266 with MQTT using this library, https://github.com/Imroy/pubsubclient

Both of these have there own PubSubClient.h , so cannot be in my arduino library at the same time !

My problem is when I compile my code for Ethernet Arduino it will only compile error free using the knolleary library and the ESP8266 will only compile error free using the Imroy library.

This gives me the problem of removing one then installing the correct library every time I use ESP8266 or Ethernet Arduino code.

How can I get both to work ? 

regards

Gary

Imroy, 2015-06-22:
The problem seems to revolve around my library including from WiFiClient.h, which is provided by either ESP8266WiFi or the Arduino WiFi library. This is because early in my customising of pubsubclient I didn't like that the constructor took a "Client" object from the sketch. I thought "the sketch isn't doing anything else with that object, pubsubclient can handle that itself". I didn't realise what effect this would have on compatibility.
So it looks like I'll have to go back to having the sketch passing a "Client" object to the constructor.

ichilton, 2015-06-30:
Is it possible to rename your library, so it can co-exist with the standard knolleary/pubsubclient?

Calling this one something like esp8266_pubsubclient wuold be much better.

Thanks,

Ian

Testato, 2015-06-30:
+1
This is a very good idea

>
> Is it possible to rename your library, so it can co-exist with the
standard knolleary/pubsubclient?

Imroy, 2015-07-06:
I made changes so that it could work with any platform providing a "Client" interface (WiFiClient, EthernetClient, etc). There's nothing really ESP8266 specific that I know of, so calling it esp8266_pubsubclient would not make sense.
Although, calling it "pubsubclient2" might be justifiable.

questuk, 2015-07-28:
Hi Imroy,

Just wondered if you have got any further on getting the library working for Ethernet Arduino board and  ESP2866 module ?

Regards

Gary