Imroy/pubsubclient #20: When including client.loop() in loop(), esp8266 reboots.

CptanPanic, 2015-07-18:
So I have the code working, and I was sending the uptime to the mqtt server, but was only getting to 25secs before restarting.  I tried changing the number of times I sent out mqtt stuff, but it didn't change anything.  Turns out if I removed client.loop() where client is a PubSubClient, it doesn't reboot anymore.  So any ideas?  Also if I am not using any callbacks do I even need to call PubSubClient::loop()?

Imroy, 2015-07-18:
Yes, the client.loop() method pings the broker regularly. So it's good for keeping the connection up.

How recent is your checkout of pubsubclient? There was issue #15 that was fixed almost a month ago. And I made some changes yesterday. So this could be a new or old problem :)

CptanPanic, 2015-07-18:
I pulled repo at about 22:30 UTC yesterday

Imroy, 2015-07-18:
Okay. A few of my changes yesterday broke the library. I spent today tracking down my stupid errors and fixing them. I hope I didn't cause too much trouble.

Testato, 2015-07-18:
Cptan, try whit the 1.99 version, the last stable release, it is about 25day old.
You found it in Releases section.
Normally the version in the releases is stable, the Master branch instead can be in development