Imroy/pubsubclient #21: .connect always returns false on NodeMCU Amica (v1.0?)

dantheman2865, 2015-07-20:
I spent some time trying the examples (mqtt_basic and mqtt_subscriber) and I could never get my device to register to the Mosquitto broker I'm running on a local server, that is to say .connect always returns false. I have verified that I get an IP, and I can see traffic going to and from the server.

I have a NodeRed instance running and I've never seen $SYS/broker/clients/total go above 1 (which is the NodeRed MQTT client).

I ran Wireshark on the server and here's the packets I see back and forth for each loop:
client to server: SYN
s to c: SYN, ACK
c to s: ACK
c to s: PSH, ACK (CONNECT message)
s to c: ACK
s to c: FIN, ACK
c to s: RST, ACK
s to ?: ACK

Here's the CONNECT message in wireshark:
`0000   10 19 00 04 4d 51 54 54 04 02 00 0f 00 0d 61 72  ....MQTT......ar`
`0010   64 75 69 6e 6f 43 6c 69 65 6e 74 xx xx xx xx xx  duinoClient`

What else do you need to troubleshoot?

Thanks for your work, I'm looking forward to getting this working!

Imroy, 2015-07-20:
What version of Mosquitto are you using?

dantheman2865, 2015-07-20:
I'm running 0.15-2ubuntu1
It says that's a v3.1 broker, is that the problem? Let me try running the Mosquitto Debian repository rather than the Ubuntu repo.

dantheman2865, 2015-07-20:
Success! I solved this by updating my broker to a v3.1.1 broker (1.4.2a) that was in the ppa repository.
Thanks for your help.

closed 2015-07-20 by dantheman2865