Imroy/pubsubclient #22: client.loop() introduces huge delays

mactro, 2015-07-22:
I have following code running on ESP:
```
void loop()
{
	if(millis() - t  > 10) {
		static int i = 0;
		String topic = "test/" + String(i % 10);
		String msg = "some test message " + String(t);
		mqtt->publish(topic, msg);
		i++;
		t = millis();
	}	
  mqtt->loop();	
}
```
When I subscribe to the test/# topic I would expect the counter on the end of the message to have difference close to 10 between subsequent messages, but I usually get something around 100-200. What is even more bizzare, when I set up a python script on the PC to publish messages in 30ms intervals to a topic that ESP is subscribed to, the publishing rate accelerates to something around 20ms. Unfortunately after some time, ESP reboots with 
```
rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 28780, room 16 
tail 12
chksum 0x35
ho 0 tail 12 room 4
load 0x3ffe8000, len 1548, room 12 
tail 0
chksum 0x90
load 0x3ffe8610, len 3300, room 8 
tail 12
chksum 0x88
csum 0x88
```

Testato, 2015-07-22:
Mqtt is writed for realtime execution ?
Example, is it writed for drive a Drone ?
I think about it like a notify sistem

mactro, 2015-07-22:
Of course mqtt is not for real time, but publishing or receiving 100 messages per second is faaar from real time. Besides, I don't mind occasional delays, as long as average frequency will be around these 100Hz.

Imroy, 2015-07-23:
To see how long the publish() and loop() methods can take, I modified the mqtt_basic sketch like this:
```C++
  uint32_t t = millis();
  client.publish("outTopic", "hello world");
  Serial.printf("Publish took %dms\n", millis() - t);
```
and
```C++
  t = millis();
  client.loop();
  Serial.printf("Loop took %dms\n", millis() - t);
```
Client.loop() almost always took "0ms" (i.e < 1 ms). Client.publish() took much longer - from less than 10 ms to over 100 ms. I don't know if there's much we can do about this. It's all up to the Espressif SDK and the Wifi network.

mactro, 2015-07-23:
Ok, I have edited a topic to reflect your measurements. I have also done some additional testing with different QoS settings.  
* QoS 0 gives results as desribed in the first post
* QoS 1 takes around 50ms between loop iterations, but there are 10 duplicate messages for every single message published (so 11 same msgs in total). 
* With QoS 2, I can run the loop at 100Hz rate, but ESP reboots after couple of seconds.

sudheera8, 2017-01-16:
Hi All

I  have been using this library with ATWINC1500. I started off with using  knolleary/pubsubclient
 but my requirement for QOS2 made me this fork of it. However,I have been experiencing the same speed issues. Can any one help me figure this out?
Thanks in advance.

sudheera