Imroy/pubsubclient #25: mosquitto v1.3 on ubuntu 14 does not accept mqtt header

mattvenn, 2015-07-23:
Might be worth changing the readme to reflect this?

mosquitto[21836]: Invalid protocol "MQTT" in CONNECT from 192.168.0.7

mattsmac:2107: /usr/sbin/mosquitto 
1437687094: mosquitto version 0.15 (build date 2013-08-23 19:23:43+0000) starting
1437687094: Opening ipv4 listen socket on port 1883.

mattsmac:2109: cat /etc/issue
Ubuntu 14.04.2 LTS \n \l

mattvenn, 2015-07-23:
btw, thanks for the lib - it works great after updating mosquitto!

matbor, 2015-07-26:
```1437687094: mosquitto version 0.15 (build date 2013-08-23 19:23:43+0000) starting```

Looks like you are running an old version 0.15 of Mosquitto, ubuntu repo is normally pretty old.

Please see the instructions on http://mosquitto.org/download/ for Ubuntu for adding the PPA and you will get the latest version, 1.4.2, should work fine with this version as the older ones aren't supported.