Imroy/pubsubclient #27: can't connect with String hostname

Sushinn, 2015-07-25:
I've set up a DDNS on my router and I'm tring to connect the ESP with your library using the constructor client(wclient, server) where server is a String with my hostname, Ex. "mybroker.ddns.net".

However this does not work.

Imroy, 2015-07-27:
Hmm, I'm seeing this too. I'll see what I can find.

Sushinn, 2015-07-31:
@Imroy maybe I could use the mDNS address of my broker (a raspberry pi), raspberrypi2.local, whould this work?

I'll try as soon as possible

Sushinn, 2015-07-31:
Solved the problem declaring the hostname as a char array, not a string. (Ex.: char server[] = "mybroker,ddns.net")

However, I can't do the same with the mDNS/Bonjur address of my broker. Maybe a mDNS resolver is necessary?

closed 2020-05-31 by marciogranzotto