Imroy/pubsubclient #30: ESP restarting when publishing

sticilface, 2015-08-04:
I got a situation where the ESP reboots when publishing any message, restarting the mosquito server fixed the problem immediately.

I have lots of the same binaries running (6 ESPs) so it is not the firmware.  

Is there a way to enable debugging of the pubsubclient to work out what the server is sending back on a publish that would cause the ESP to reboot, should this happen again? Then we might be able to implement a fix to stop the ESP rebooting under these circumstances? 

Cheers

A

Sushinn, 2015-08-04:
You could use some tool like http://www.jensd.de/apps/mqttfx/ to monitor the broker's activity. Subscribe to all topic that the ESP is subscribing and try to figure out what is wrong

sticilface, 2015-08-05:
I'm using MQTTspy which from a quick glance does the same thing.  I was thinking that i needed a closer look at exactly what what being exchanged between the device and the broker..

Testato, 2015-08-05:
Are you using the last Stable ESP Arduino Core version ?
It solve some auto restart bug.

sticilface, 2015-08-05:
Pretty much.

I just did a git pull  
```
Updating fd1d9da..5f99bf8
```

the only files changed were ESPTOOL, and some FS files, and documents.

Testato, 2015-08-05:
Dorry i'm on a cellphone so a can not control easly, but is there two
version and two different Board Manager json link.
The development one is not stable, it can become broken every day because
it is for development test.
So you are tried whith the Stable Json link ?

Imroy, 2015-08-05:
I've used WireShark to capture and view the exchange of messages. It doesn't decode the packets by default though. You have to right-click on a packet, select "Decode as..." from the menu, then select MQTT from the list of protocols.

sticilface, 2015-08-05:
Sounds like a good idea.  I'll see what i can do next time it does it