Imroy/pubsubclient #31: Receives it's own message

hahathejellyfox, 2015-08-05:
Hello, I have been really enjoying this library however I've hit a dead end in my code. My device needs to send and receive messages on the same topic but when I publish a message, it also receives the same message.

I was wondering if there is a workaround for this issue or if it is a issue with the library itself.

Imroy, 2015-08-05:
I think that's a valid behaviour of the protocol. When you subscribe to a topic pattern, you receive all messages that match, even your own.

Can you put a node id (MAC address or ESP.getChipId()) in either the topic or payload to make it possible to filter out your own messages?

hahathejellyfox, 2015-08-05:
I guess it makes sense for it to work like that, for now I have created a temporary variable that gets set to the message payload and is checked against incoming messages in the callback function, if it matches I discard the message. This should work fine for my application but your solution is a lot more sturdier so I'll try that out.

Cheers for the help! :)

closed 2015-08-05 by hahathejellyfox