Imroy/pubsubclient #35: OTA issues

mtnbrit, 2015-09-18:
Hi  Iml having trouble getting OTA to work reliably over MQTT

I'm using @Imroy 's MQTT example as a base for my code, it uses UpdaterClass::writeStream

I added a lot of extra debugging to Updater.cpp to see why it was failing 98% of the time. I found the mqtt transfer stream was "bogging down", i.e. it starts out fast but gets slower and slower and eventually times out. I added ```data.setTimeout(15000);``` to the top of writeStream and this helped but I still get timeouts. This is from a broker located in a Chicago hosting co, so only about 70ms away and no packet loss. 

In order to eliminate network conditions as being the cause, I tried also from a broker on my local LAN, i.e. less than 1ms latency, and things look much better, i can get 6 out of 10 OTAs to work perfect. I get the "Staged" message and it restarts as it should.

Im getting lmac.c 662 the other 4 times out of 10 when testing from the LAN broker. 

Back to testing from my internet-hosted broker (which is ultimately where it needs to work from), it mostly just bogs down and times out with ```ERROR[5]: Stream Read Timeout
```, sometimes I get lmac.c 662 also right after starting the OTA. The packets start out coming in at 4096 bytes each nice and rapidly, but they seem to slow down i.e. they get longer and longer gaps between them, eventually I start seeing some smaller-than-4096 packets, this is a pretty good sign its about to timeout, then it does.  I added ```Serial.println(toRead);
Serial.println(written);``` in the while inside writeStream to view the progress, heres the output

```
.
.
.
4096
143360
4096
147456
4096
151552
4096
155648
4096
159744
4096
163840
4096
167936
4096
172032
4096
176128
4096
180224
4096
184320
4096
188416
4096
192512
4096
196608
4096
200704
1026
204800
1460
205826
0
207286
ERROR[5]: Stream Read Timeout
ERROR[5]: Stream Read Timeout

```

There is no packet loss from the esp to my gateway and no loss from the gateway to the server, no congestion anywhere. It almost feels like my server is rate-limiting or throttling this large mqtt transfer, is that even possible? I'm running mosquito 1.4.3 broker. 

Any ideas?

Sushinn, 2015-09-18:
Is MQTT really suitable for large transfers like this? What I would do is just send a MQTT message to notify the ESP that a OTA is available on some server and then the ESP would get the update from there. 

Again, I don't know what I'm saying, that's just what I would do.

Testato, 2015-09-18:
Are your code opensource ? I need ota uodate also ;-)

But i agree whit Marcio, the official SDK have OTA methods and if i
remember the  esp core for Arduino also, so why create a new type of OTA on
Mqtt protocoll ?

emre-cakir, 2021-12-08:
@mtnbrit Hi , i have same problem, is this problem fixed ? please help :)