Imroy/pubsubclient #36: Another OTA Issue

brutevinch, 2015-11-02:
Using last version of lib.

Can't compile example OTA

METEOSTATION.ino: In function 'void receive_ota(const MQTT::Publish&)':
METEOSTATION:57: error: 'const class MQTT::Publish' has no member named 'payload_stream'
METEOSTATION:58: error: 'const class MQTT::Publish' has no member named 'payload_stream'
'const class MQTT::Publish' has no member named 'payload_stream'

Imroy, 2015-11-09:
It looks like you have an old version of pubsubclient installed.

brutevinch, 2015-11-09:
@Imroy Can you help me with another problem? After update OTA i've got loop.... i think problem in broker but i can't understand it.

Connecting to AirPort Extreme 2,4
***
WiFi connected
Hostname: meteostation.local
IP address: 10.0.1.102
Local Pressure: 750.9
Connecting to MQTT server
Connected to MQTT server
Subscribing to topic ota/10.0.1.102
Receiving OTA of 335312 bytes...
Clearing retained message.
Update Success: 12115
Rebooting...
rm 0
pm close 7 0 0/14577359
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0
ld

Connecting to AirPort Extreme 2,4
*******
WiFi connected
Hostname: meteostation.local
IP address: 10.0.1.102
Local Pressure: 750.6
Connecting to MQTT server
Connected to MQTT server
Subscribing to topic ota/10.0.1.102
Receiving OTA of 335312 bytes...
Clearing retained message.
Update Success: 12062
Rebooting...
rm 0
pm close 7 0 0/14526063
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0
ld

igrr, 2015-11-09:
This looks like two successful OTAs to me. What's the issue?

brutevinch, 2015-11-09:
@igrr YES but it in looop!!! I need using device not only update OTA )))))))))))

Imroy, 2015-11-10:
Have you only sent the OTA message once? The sketch should be clearing the retained message once it has successfully flashed the new image.

brutevinch, 2015-11-10:
@Imroy Yes i sent only 1 message and when i stop mosquito broker problem is gone. i close this issue. i will use igrr ota. Thank you

closed 2015-11-10 by brutevinch