Imroy/pubsubclient #40: Is it possible to connect to two MQTT brokers at the same time with this library?

edautz, 2015-12-04:
From a friend i got an ESP sketch to get P1 readings form an powermeter and send them to an Mosquitto MQTT broker.

It works well.

I try to change the sketch so i can connect to two Mosquitto MQTT brokers at the same time, but I endup in a connection loop when using the  client1.loop(); and client2.loop() statements.

Does this library support using two clients simultaniously?

Imroy, 2015-12-05:
Well it's not been tested doing that. But everything is nicely encapsulated and there are no global variables, so it shouldn't be a problem.

How are you calling client1.loop() and client2.loop(), and just what is this "connection loop"?

edautz, 2015-12-05:
After publish some data to the MQTT broker, the sketch calls client1.loop() or client2.loop(). 

While calling it, it generates lots of connections to the broker. 

I can send the sketch if this helps.

matbor, 2015-12-06:
As a work around, if you not aware, you can setup one of your brokers in bridge mode so that you can publish the information to the second broker.  http://mosquitto.org/man/mosquitto-conf-5.html

edautz, 2015-12-07:
I am aware of that solution, but I got two seperate MQTT brokers and two seperate homedomtica systems for redundancy and testing purposes. I want to keep my setup this way.