Imroy/pubsubclient #43: QOS 1 Puback

MikeSchafer, 2015-12-28:
When the client receives a qos = 1 publish from the server, it does not include the packet ID or the return code in the puback response.

MikeSchafer, 2015-12-30:
Correction -- There is no response code in a puback, just the packet ID of the received packet.

Imroy, 2017-03-19:
Sorry for not dealing with this sooner. THKDev just fixed this :P

closed 2017-03-19 by Imroy

MikeSchafer, 2017-03-19:
Thanks.

Mike

----------------------------------------------------------------------------------------------------------
R. Michael Schafer, Ph.D.                     E-mail: RMSchafer@nd.edu
Teaching Professor                              Phone: (574) 631-4851
273 Fitzpatrick HL ENGRNG
Department of Electrical Engineering
University of Notre Dame
Notre Dame, IN 46556