Imroy/pubsubclient #47: How Clean Session work?

r666t, 2016-01-29:
Hi lmroy,
I'm trying my esp retrieve messages from the server that may have existed while he was off. I tried to connect with the variable:
- Set_clean_session (true)
- Set_clean_session (false)
- Unset_clean_session ()
but I can not get my esp recover lost messages.
Must I configure anything something on the mosca server?

Thanks and Regards

Wonderfull library!!!!!!!!!!!!!

sudheera8, 2017-04-19:
Hi @Imroy , @r666t 

I am having the same issue. How do we use the clean session flag to get the messages sent when the device was offline. Please respond. This is critical for my application. 

Thanks
Sudheera

robynjayqueerie, 2017-08-15:
I also struggled with this but I think Imroy answers it
See https://github.com/Imroy/pubsubclient
and go down to
Setting options on messages

The PubSubClient class operates mostly as it did before. However, the connect(), publish(), subscribe(), and unsubscribe() methods can now take an appropriate MQTT object. This allows extra options to be set e.g QoS on publish, or multiple topics with one (un)subscribe message.

You can use the MQTT classes and their chainable setter methods like this:

 client.connect(MQTT::Connect("clientId")
                .set_clean_session()
                .set_will("status", "down")
                .set_auth("username", "password)
                .set_keepalive(30)
               );
Basically you need to use the unset_clean_session() here