Imroy/pubsubclient #48: Exception (9) and (28) with 2.1.0-rc2, but not with 2.1.0-rc1

mtnbrit, 2016-02-02:
Hi im getting random exception 28 and exception 9 when doing a simple client.publish(), but only with 2.1.0-rc2, with 2.1.0-rc1 it doesn't fail.

perhaps @igrr would be interested in this issue too.

here are the crashes

```
loop: 53
loop pub ready
Fatal exception 28(LoadProhibitedCause):
epc1=0x40105cb2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00228000, depc=0x00000000

Exception (28):
epc1=0x40105cb2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00228000 depc=0x00000000

ctx: cont 
sp: 3fff2c80 end: 3fff3200 offset: 01a0

>>>stack>>>
3fff2e20:  5c5c5c5c 5c5c5c5c 5c5c5c5c 5c5c5c5c  
3fff2e30:  c20b8380 0000016e 40248348 40227286  
3fff2e40:  00000000 00000000 000005b4 97205237  
3fff2e50:  0000005a 36363636 4022a56c 3fff4458  
3fff2e60:  00000000 00000000 0d2c3dde 2683964c  
3fff2e70:  00000055 3fff2e90 4022a78d 08080808  
3fff2e80:  00000001 9693819c 0d2c3dde 2683964c  
3fff2e90:  ade50000 00000000 00000000 4c968326  
3fff2ea0:  ade5e780 9c819396 3fff61e8 00000000  
3fff2eb0:  00000001 402078a3 3fff4510 00000000  
3fff2ec0:  00000055 00000055 00000055 00000055  
3fff2ed0:  00000000 3fff5964 00000000 00000001  
3fff2ee0:  00000055 2683964c 3ffea32a 00000050  
3fff2ef0:  3fff47d8 00000000 00000055 40236b88  
3fff2f00:  00020317 40104c23 3ffef208 40213012  
3fff2f10:  00000023 00000017 00000010 4021306a  
3fff2f20:  0000008a beb8bcb3 3fff2f60 00000000  
3fff2f30:  3fff47d8 3fff4a28 00000023 40236c0f  
3fff2f40:  00000023 beb8bcb3 3fff3030 40212fb8  
3fff2f50:  3fff4a20 00000000 3fff3030 3fff1df4  
3fff2f60:  00000023 3fff4a28 3fff3030 402072b4  
3fff2f70:  00000023 3fff4a28 3fff3030 4020dde7  
3fff2f80:  00000023 40102a84 3fffc200 00000022  
3fff2f90:  00000021 00000000 3ffea32a 4010754c  
3fff2fa0:  3fff3054 0000001d 00000020 00000000  
3fff2fb0:  3fff1db4 3fff3030 00000001 4020e78c  
3fff2fc0:  00000000 00000000 00000001 4021306a  
3fff2fd0:  00000005 00000000 3fff3054 3fff1db4  
3fff2fe0:  3fff3030 3fff1db4 00000001 4020e955  
3fff2ff0:  3fff3030 3fff306c 00000000 402131c0  
3fff3000:  3fff4a20 40107563 3fff3078 4020def5  
3fff3010:  0000000c 3fff47d8 00000001 40212fb8  
3fff3020:  3fff30b0 3fff1db4 00000001 4020e9e5  
3fff3030:  3ffea478 00000003 0000a300 40107500  
3fff3040:  00000000 00000002 00000010 00000000  
3fff3050:  3fff2118 3fff4a90 0000001f 0000001d  
3fff3060:  3fff4ab8 00000002 3fff3001 00000000  
3fff3070:  00000000 00000000 00000000 00000000  
3fff3080:  00000000 3fff1ec8 3fff30b0 402130e8  
3fff3090:  3fff3178 ff000000 3fff2118 40212734  
3fff30a0:  3fff18ec 3fff1ec8 3fff2118 4020581f  
3fff30b0:  3fff3c70 0000000f 00000002 401058c9  
3fff30c0:  4000050c beb72cfa 00000000 4000050c  
3fff30d0:  00000000 00000000 0000001f 401058c9  
3fff30e0:  4000050c 00000023 00000001 00000001  
3fff30f0:  40210a58 00000030 0000000a ffffffff  
3fff3100:  40100114 3fff21e0 40210a50 00000000  
3fff3110:  00000000 6e696874 312f7367 65663a38  
3fff3120:  3a34333a 353a6239 31383a62 6f6f6c2f  
3fff3130:  00000070 3fffdad0 3fff21d8 00000030  
3fff3140:  00000000 3fffdad0 3fff21d8 00000030  
3fff3150:  3fff2118 00000008 3fff2118 40210165  
3fff3160:  3ffe9ac2 00000001 3fff2118 40210165  
3fff3170:  3ffe9438 3fff21d8 3fff4a00 0000001f  
3fff3180:  0000001d 00000000 00000000 00000000  
3fff3190:  40003335 00000001 3fff18ec 4020f3d8  
3fff31a0:  3fffdad0 3fff1914 3fff18ec 40205158  
3fff31b0:  00000000 00000000 00000000 00000000  
3fff31c0:  40203ff8 feefeffe 40210c8c 40210c78  
3fff31d0:  00000000 00000000 00000001 3fff21d8  
3fff31e0:  3fffdad0 00000000 3fff21d1 40210a78  
3fff31f0:  feefeffe feefeffe 3fff21e0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

and 

```
loop: 9
loop pub ready
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x40105b29, epc2=0x00000000, epc3=0x00000000, excvaddr=0x80fb0f6d, depc=0x00000000

Exception (9):
epc1=0x40105b29 epc2=0x00000000 epc3=0x00000000 excvaddr=0x80fb0f6d depc=0x00000000

ctx: cont 
sp: 3fff2c80 end: 3fff3200 offset: 01a0

>>>stack>>>
3fff2e20:  5c5c5c5c 5c5c5c5c 5c5c5c5c 5c5c5c5c  
3fff2e30:  5c5c5c5c 5c5c5c5c 40248348 40227286  
3fff2e40:  00000000 00000000 000005b4 4cfc6cb5  
3fff2e50:  0000005a 36363636 4022a56c 3fff3f70  
3fff2e60:  00000000 00000000 0864472e 774d269b  
3fff2e70:  00000055 3fff2e90 4022a78d 09090909  
3fff2e80:  00000001 5c741514 0864472e 774d269b  
3fff2e90:  a4640000 00000000 00000000 9b264d77  
3fff2ea0:  a464c1d5 1415745c 3fff5d50 00000000  
3fff2eb0:  00000001 402078a3 3fff40f0 00000000  
3fff2ec0:  00000055 00000055 00000055 00000055  
3fff2ed0:  00000000 3fff560c 00000000 00000001  
3fff2ee0:  00000055 774d269b 3ffea32a 00000050  
3fff2ef0:  3fff4240 00000000 00000055 40236b88  
3fff2f00:  00020317 40104c22 3ffef208 40213012  
3fff2f10:  00000022 00000017 00000010 4021306a  
3fff2f20:  0000008c 1cca2a58 3fff2f60 00000000  
3fff2f30:  3fff4240 3fff46b0 00000022 40236c0f  
3fff2f40:  00000022 1cca2a58 3fff3030 40212fb8  
3fff2f50:  3fff46a8 00000000 3fff3030 3fff1df4  
3fff2f60:  00000022 3fff46b0 3fff3030 402072b4  
3fff2f70:  00000022 3fff46b0 3fff3030 4020dde7  
3fff2f80:  00000022 00000000 0000001f 401058c9  
3fff2f90:  00000020 00000000 3ffea32a 4010754c  
3fff2fa0:  3fff3054 0000001d 00000020 00000000  
3fff2fb0:  3fff1db4 3fff3030 00000001 4020e78c  
3fff2fc0:  00000000 00000000 00000001 4021306a  
3fff2fd0:  00000009 00000000 3fff3054 3fff1db4  
3fff2fe0:  3fff3030 3fff1db4 00000001 4020e955  
3fff2ff0:  3fff3030 3fff306c 00000000 402131c0  
3fff3000:  3fff45c0 40107563 3fff3078 4020def5  
3fff3010:  0000000c 3fff4240 00000001 40212fb8  
3fff3020:  3fff30b0 3fff1db4 00000001 4020e9e5  
3fff3030:  3ffea478 00000003 0000a300 40107500  
3fff3040:  00000000 00000001 00000010 00000000  
3fff3050:  3fff2118 3fff4688 0000001f 0000001d  
3fff3060:  3fff45f0 00000001 3fff3001 00000000  
3fff3070:  00000000 00000000 00000000 00000000  
3fff3080:  00000000 3fff1ec8 3fff30b0 402130e8  
3fff3090:  3fff3178 ff000000 3fff2118 40212734  
3fff30a0:  3fff18ec 3fff1ec8 3fff2118 4020581f  
3fff30b0:  3fff38f8 0000000f 00000001 401058c9  
3fff30c0:  4000050c 00000000 0000001f 401058c9  
3fff30d0:  40000f83 00000030 0000000d ffffffff  
3fff30e0:  40000ea3 00000023 00000001 00000001  
3fff30f0:  3fffdab0 00000000 3fffd9d0 3fff21d8  
3fff3100:  00000000 00000000 00000001 3fff21d1  
3fff3110:  00000000 6e696874 312f7367 65663a38  
3fff3120:  3a34333a 353a6239 31383a62 6f6f6c2f  
3fff3130:  00000070 3fffdad0 3fff21d8 00000030  
3fff3140:  00000000 3fffdad0 3fff21d8 00000030  
3fff3150:  3fff2118 00000008 3fff2118 40210165  
3fff3160:  3ffe9ac2 00000004 3fff2118 40210165  
3fff3170:  3ffe9438 3fff21d8 3fff45a0 0000001f  
3fff3180:  0000001d 00000000 00000000 00000000  
3fff3190:  40200039 00000004 3fff18ec 4020f3d8  
3fff31a0:  3fffdad0 3fff1914 3fff18ec 40205158  
3fff31b0:  00000000 00000000 00000000 00000000  
3fff31c0:  40203ff8 feefeffe 40210c8c 40210c78  
3fff31d0:  00000000 00000000 00000001 3fff21d8  
3fff31e0:  3fffdad0 00000000 3fff21d1 40210a78  
3fff31f0:  feefeffe feefeffe 3fff21e0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

```

this is my code

```
    loops++;
    DEBUG.println(); 
    DEBUG.print("loop: ");
    DEBUG.println(loops);
    char s_loops[10];
    sprintf(s_loops, "%d", loops);
    char ptopic[100];
    strcpy(ptopic, topic);
    strcat(ptopic,"loop");
    DEBUG.println("loop pub ready");
    client.publish(ptopic,s_loops);
    DEBUG.println("loop published");
```
when client.publish crashes, obviously it never gets to say "loop published"

igrr, 2016-02-02:
Could you please try the same with git version of the core? I think i have fixed one issue with lwip which could lead to these crashes after rc2.

mtnbrit, 2016-02-03:
It looks good so far, thanks @igrr.

closed 2016-02-03 by mtnbrit