Imroy/pubsubclient #51: ESP 8266 hangs when the incoming message rate is too high.

HW-Siew, 2016-03-07:
Hi, all. I am experiencing a random hangs with ESP8266 when the pubsubclient incoming message rate is too high (around 10-50msg/s). I found a potential cause [https://github.com/esp8266/Arduino/issues/50](url). Any idea how we can fix this. Tq.

tijnkooijmans, 2016-03-17:
I have the same issue, any progress on this?

HW-Siew, 2016-03-18:
Hi, this still remains unsolved. However, there are few possible solutions as mentioned in [https://github.com/esp8266/Arduino/issues/50](url). I do not have time to test it yet. if you do, do let me know how does it work. Tq

tijnkooijmans, 2016-03-26:
It seems putting a 'loop()' call in front of every publish/subscribe call works. I use this fork (https://github.com/Suxsem/pubsubclient) that has a fix for it.

git-tiger, 2017-12-18:
I am sending UDP commands at frequency of 50Hz. After a period, the ESP hangs.