Imroy/pubsubclient #71: Wrong password

marciogranzotto, 2017-01-08:
I'm using a broker that requires an username and password. However, if I set the password wrong in `clientMQTT.connect(MQTT::Connect(clientID).set_auth(conf.mqttUser, conf.mqttPassword))` the client returns that's connected but on the next loop it's no longer connected.

On the broker side I've got a whole lot of
`1483897603: New connection from 192.168.0.192 on port 1883.
1483897603: Sending CONNACK to 192.168.0.192 (0, 5)
1483897603: Socket error on client <unknown>, disconnecting.`

My question is that if there's any way to check if the password is wrong so I can do something about it on my code.

lawrence-jeff, 2018-03-02:
Similar question here - would like to pickup the ConnectAck value of 4 which according to the spec should indicate incorrect username and password - not seeing how to get the return value .. tried looking at the cpp and I see reference to ConnectAck being a class but can't figure out how to catch it

Imroy, 2018-03-03:
I've just pushed an update. It's work I did in May last year but didn't get around to pushing to github - I hadn't tested it on actual hardware and was afraid of breaking peoples' projects (even though a lot of the work involved getting the old testing framework working again).

Anyway, one of the changes I made was that PubSubClient::connect() returns false if the return code in the CONNACK packet doesn't equal zero. If you really need the rc value I could add that.

lawrence-jeff, 2018-03-03:
Great - thanks! I will test that and I can probably make that work. If your curious on the use case I have a project where the current code only supports anonymous connections but there is a request to also support U/P. There is already the concept of an ID/pwd for other purposes so my thought was I could just update the code to try anonymous first and if it gets rc 5 (what Mosquito at least sends when password is required) then catch that and retry with the existing username and password. That way no existing anonymous users have to change anything and I don't have to update the configuration UI with options around this mqtt security. Anyway appreciate the work, I really wish I was more proficient in c/c++ or I would try to help and submit a PR - I did notice the project you forked from has rc support if you ever do want to add it.
Thanks again and will test it out tonight and let you know