Imroy/pubsubclient #72: lib update installs the knolleary lib

consolacion, 2017-01-14:
Oddly, if the Arduino library manager wants to update this library, it installs the original knolleary library rather than this fork. Cant understand why

Testato, 2017-01-14:
because arduino need that all library have a different name in the porprierty file.
, i will create a PR for solve this Issue

consolacion, 2017-01-14:
ok. Still surprised though that it starts updating to knolleary lib as nowhere in my setup there is any reference to knolleary file :-)

Testato, 2017-01-14:
but on your machine you have the imroy library installed, so the integrated arduino library update system think that you are using the knolleary library, because the imroy proprierty file is the only file that arduino IDE check. It do not check the library folder name, or the library constructor name, etc etc.

Testato, 2017-01-14:
but on your machine you have the imroy library installed, so the integrated arduino library update system think that you are using the knolleary library, because the imroy proprierty file is the only file that arduino IDE check. It do not check the library folder name, or the library constructor name, etc etc.

Pay attention that this do not mean that you will receive automatically imroy update, because knolleary is present in the Arduino update lib management, instead this lib is not present in it at moment. see this my issue ;-)
https://github.com/Imroy/pubsubclient/issues/19

consolacion, 2017-01-14:
ok, maybe solution then to change the name. thanks for great work